mwarning/openwrt-firmware-selector #2: Change database to Names and IDs only

aparcar, 2020-02-10:
I created a script that uses the `tmp/.targetinfo` file to generate a file called `names.json` looking like below. This could be used to load specific device information on demand instead of all together. Using `target` and `id` can be combined as `{ release }/{ target }/{ id }.json`. 

Would you be willing to change the current script to that?

```
{
  "FriendlyARM NanoPi NEO2": {
    "target": "sunxi/cortexa53",
    "id": "friendlyarm_nanopi-neo2"
  },
  "Pine64 Pine64+": {
    "target": "sunxi/cortexa53",
    "id": "pine64_pine64-plus"
  },
  "Pine64 SoPine": {
    "target": "sunxi/cortexa53",
    "id": "pine64_sopine-baseboard"
  },
  "Xunlong Orange Pi PC 2": {
    "target": "sunxi/cortexa53",
    "id": "xunlong_orangepi-pc2"
  },
  "Xunlong Orange Pi Zero Plus": {
    "target": "sunxi/cortexa53",
    "id": "xunlong_orangepi-zero-plus"
  },
  "CompuLab TrimSlice": {
    "target": "tegra",
    "id": "compulab_trimslice"
  },
  "Generic x86": {
    "target": "x86/generic",
    "id": ""
  },
  "Generic x86/legacy": {
    "target": "x86/legacy",
    "id": ""
  },
  "Geos": {
    "target": "x86/geode",
    "id": ""
  },
  "Generic x86/64": {
    "target": "x86/64",
    "id": ""
  },
  "Avnet ZedBoard": {
    "target": "zynq",
    "id": "avnet_zynq-zed"
  },
  "Digilent Zybo": {
    "target": "zynq",
    "id": "digilent_zynq-zybo"
  },
  "Digilent Zybo Z7": {
    "target": "zynq",
    "id": "digilent_zynq-zybo-z7"
  },
  "Xilinx ZC702": {
    "target": "zynq",
    "id": "xlnx_zynq-zc702"
  }
}
```

mwarning, 2020-02-10:
Why use tmp/.targetinfo as another source when the json files are available as well?

mwarning, 2020-02-10:
A for splitting up the data in a quick to load json - that might be a good idea. But I like to keep things simple and the current size of the data does not seem to require this (the data.json is only <40KB compressed).

mwarning, 2020-02-10:
Your proposed format is ok, but we would need to extend that to different releases.

aparcar, 2020-02-10:
> Why use tmp/.targetinfo as another source when the json files are supposed to be available as well?

Using targetinfo allows you to get all available models without building all. When you want to parse the online JSON files, then you also have to download about 1500 json files with details to get the model names.

However it has the disadvantage that we need to do this for every distro version we want to support. So run `<script>.py` for master, 19.07, etc ...

The JSON files would actually be called `%distro-%version-%target-%profile.json`, where `%version` is skipped on master builds and `%target` uses `-` instead of `/`.

mwarning, 2020-02-12:
@aparcar I am more or less finished with the restructuring. But

- I cannot use %target to mean ../.. in one place, and ..-.. in another place. I could split target into %target and %subtarget. But then %target would have two conflicting definitions. => we need a different variable name. Ideas?
- Your example JSON uses 'id', but you also call it 'profile'. We should only use one name. Which one?

mwarning, 2020-02-12:
Hm, but when I would define something like %maintarget-$subtarget in the link, and $subtarget is empty, then the link would break, because the - would not supposed to be there.

mwarning, 2020-02-12:
@aparcar I changed the json format to something yet simpler. Feel free to submit a PR to change it.. :-)

closed 2020-02-17 by mwarning