mwarning/openwrt-firmware-selector #10: Use Bootstrap 4.x

aparcar, 2020-02-19:
This makes the website also usable via mobile devices

mwarning, 2020-02-19:
It is probably easy to do even without bootstrap. Just say'in. :p

aparcar, 2020-02-19:
I think LuCI as well as the OpenWrt websites moves to Bootstrap, so no time wasted in learning that :)

aparcar, 2020-02-20:
[image]

I worked a bit on it but maybe it is to much overhead. @mwarning if your up please integrate the new textarea/button in your current css and we skip bootstrap...

mwarning, 2020-02-20:
I will give it a try.

aparcar, 2020-02-21:
We still have to make it mobile friendly...

mwarning, 2020-06-16:
It makes sense to use bootstrap if the design gets more complicated.
But I think that we should limit the project to basic and easy to implement features.

More complex scenarios are better served with a different project / fork of YAFS..

closed 2020-06-16 by mwarning