mwarning/openwrt-firmware-selector #15: Show packages of custom images

aparcar, 2020-02-25:
Custom images offer a `manifest.json` which contains a dict of package and version. An example is found [here](https://chef.libremesh.org/store/SNAPSHOT/ramips/mt7621/xiaomi_mir3p/c4f0d04e275d/manifest.json). This should be viewable in the front-end.

aparcar, 2020-02-25:
The `manifest` is now part of the build request method:

```
{
  "bin_dir": "SNAPSHOT/ipq40xx/generic/avm_fritzbox-4040/4bb3b9925028",
  "build_at": "Tue, 25 Feb 2020 09:40:55 GMT",
  "buildlog": true,
  "enqueued_at": "Tue, 25 Feb 2020 09:40:39 GMT",
  "id": "avm_fritzbox-4040",
  "image_prefix": "openwrt-4bb3b9925028-ipq40xx-generic-avm_fritzbox-4040",
  "images": [
    {
      "name": "openwrt-4bb3b9925028-ipq40xx-generic-avm_fritzbox-4040-squashfs-eva.bin",
      "sha256": "a809942387212bcb6cacacaf0a94202419e598872b854c9c09436140492b118e",
      "type": "eva"
    },
    {
      "name": "openwrt-4bb3b9925028-ipq40xx-generic-avm_fritzbox-4040-squashfs-sysupgrade.bin",
      "sha256": "798e8d9ec414f4cdb6bb7c52493c488b6549469d437136e627a7624277e25a5a",
      "type": "sysupgrade"
    }
  ],
  "manifest": {
    "ath10k-firmware-qca4019-ct": "2019-10-03-d622d160-1",
    "base-files": "213-r12297-7e9c7e7b2d",
    "busybox": "1.31.1-1",
    "cgi-io": "17",
    "dnsmasq": "2.80-18",
    "dropbear": "2019.78-3",
    "firewall": "2019-11-22-8174814a-1",
    "fritz-caldata": "1",
    "fritz-tffs": "1",
    "fstools": "2020-01-21-deb745f8-1",
    "fwtool": "2019-11-12-8f7fe925-1",
    "getrandom": "2019-12-31-0e34af14-3",
    "hostapd-common": "2019-08-08-ca8c2bd2-6",
    "ip6tables": "1.8.3-2",
    "iptables": "1.8.3-2",
    "iw-full": "5.3-2",
    "jshn": "2020-01-20-43a103ff-1",
    "jsonfilter": "2018-02-04-c7e938d6-1",
    "kernel": "4.19.101-1-5732fe4c6f0fd3f52a7a96f0cda9a9f1",
    "kmod-ath": "4.19.101+5.4-rc8-1-1",
    "kmod-ath10k-ct": "4.19.101+2020-01-29-3e3d0adb-1",
    "kmod-cfg80211": "4.19.101+5.4-rc8-1-1",
    "kmod-gpio-button-hotplug": "4.19.101-3",
    "kmod-hwmon-core": "4.19.101-1",
    "kmod-ip6tables": "4.19.101-1",
    "kmod-ipt-conntrack": "4.19.101-1",
    "kmod-ipt-core": "4.19.101-1",
    "kmod-ipt-nat": "4.19.101-1",
    "kmod-ipt-offload": "4.19.101-1",
    "kmod-leds-gpio": "4.19.101-1",
    "kmod-lib-crc-ccitt": "4.19.101-1",
    "kmod-mac80211": "4.19.101+5.4-rc8-1-1",
    "kmod-nf-conntrack": "4.19.101-1",
    "kmod-nf-conntrack6": "4.19.101-1",
    "kmod-nf-flow": "4.19.101-1",
    "kmod-nf-ipt": "4.19.101-1",
    "kmod-nf-ipt6": "4.19.101-1",
    "kmod-nf-nat": "4.19.101-1",
    "kmod-nf-reject": "4.19.101-1",
    "kmod-nf-reject6": "4.19.101-1",
    "kmod-nls-base": "4.19.101-1",
    "kmod-ppp": "4.19.101-1",
    "kmod-pppoe": "4.19.101-1",
    "kmod-pppox": "4.19.101-1",
    "kmod-slhc": "4.19.101-1",
    "kmod-usb-core": "4.19.101-1",
    "kmod-usb-dwc3": "4.19.101-1",
    "kmod-usb-dwc3-qcom": "4.19.101-1",
    "kmod-usb3": "4.19.101-1",
    "libblobmsg-json": "2020-01-20-43a103ff-1",
    "libc": "1.1.24-2",
    "libgcc1": "8.3.0-2",
    "libip4tc2": "1.8.3-2",
    "libip6tc2": "1.8.3-2",
    "libiwinfo-lua": "2020-01-05-bf2c1069-1",
    "libiwinfo20200105": "2020-01-05-bf2c1069-1",
    "libjson-c4": "0.13.1-1",
    "libjson-script": "2020-01-20-43a103ff-1",
    "liblua5.1.5": "5.1.5-7",
    "liblucihttp-lua": "2019-07-05-a34a17d5-1",
    "liblucihttp0": "2019-07-05-a34a17d5-1",
    "libnl-tiny": "2019-10-29-0219008c-1",
    "libpthread": "1.1.24-2",
    "libubox20191228": "2020-01-20-43a103ff-1",
    "libubus-lua": "2020-01-05-d35df8ad-1",
    "libubus20191227": "2020-01-05-d35df8ad-1",
    "libuci20130104": "2020-01-27-e8d83732-3",
    "libuclient20160123": "2020-01-05-fef6d3d3-1",
    "libxtables12": "1.8.3-2",
    "logd": "2019-12-31-0e34af14-3",
    "lua": "5.1.5-7",
    "luci": "git-20.050.26260-d8b2e0a-1",
    "luci-app-firewall": "git-20.050.26260-d8b2e0a-1",
    "luci-app-opkg": "git-20.050.26260-d8b2e0a-1",
    "luci-base": "git-20.050.26260-d8b2e0a-1",
    "luci-lib-ip": "git-20.050.26260-d8b2e0a-1",
    "luci-lib-jsonc": "git-20.050.26260-d8b2e0a-1",
    "luci-lib-nixio": "git-20.050.26260-d8b2e0a-1",
    "luci-mod-admin-full": "git-20.050.26260-d8b2e0a-1",
    "luci-mod-network": "git-20.050.26260-d8b2e0a-1",
    "luci-mod-status": "git-20.050.26260-d8b2e0a-1",
    "luci-mod-system": "git-20.050.26260-d8b2e0a-1",
    "luci-proto-ipv6": "git-20.050.26260-d8b2e0a-1",
    "luci-proto-ppp": "git-20.050.26260-d8b2e0a-1",
    "luci-theme-bootstrap": "git-20.050.26260-d8b2e0a-1",
    "mtd": "25",
    "netifd": "2020-01-18-1321c1bd-1",
    "odhcp6c": "2019-01-11-e199804b-16",
    "odhcpd-ipv6only": "2020-01-14-6db312a6-3",
    "openwrt-keyring": "2019-07-25-8080ef34-1",
    "opkg": "2020-01-25-c09fe209-1",
    "ppp": "2.4.8-1",
    "ppp-mod-pppoe": "2.4.8-1",
    "procd": "2020-02-11-c30b23e3-1",
    "rpcd": "2020-01-05-efe51f41-2",
    "rpcd-mod-file": "2020-01-05-efe51f41-2",
    "rpcd-mod-iwinfo": "2020-01-05-efe51f41-2",
    "rpcd-mod-luci": "20191114",
    "rpcd-mod-rrdns": "20170710",
    "swconfig": "12",
    "ubi-utils": "2.1.1-1",
    "ubox": "2019-12-31-0e34af14-3",
    "ubus": "2020-01-05-d35df8ad-1",
    "ubusd": "2020-01-05-d35df8ad-1",
    "uci": "2020-01-27-e8d83732-3",
    "uclient-fetch": "2020-01-05-fef6d3d3-1",
    "uhttpd": "2020-02-12-2ee323c0-1",
    "urandom-seed": "1.0-1",
    "urngd": "2020-01-21-c7f7b6b6-1",
    "usign": "2019-09-21-f34a383e-1",
    "wireless-regdb": "2019.06.03",
    "wpad-basic": "2019-08-08-ca8c2bd2-6",
    "zlib": "1.2.11-3"
  },
  "metadata_version": 1,
  "supported_devices": [
    "avm,fritzbox-4040"
  ],
  "target": "ipq40xx/generic",
  "titles": [
    {
      "model": "FRITZ!Box 4040",
      "vendor": "AVM"
    }
  ],
  "version_commit": "r12297-7e9c7e7b2d",
  "version_number": "SNAPSHOT"
}
```

mwarning, 2020-06-12:
You wrote "build request". Does that mean that this is send to the ASU server or is it the response from the ASU server in case of a successful build?

And how should this information be displayed?

mwarning, 2020-06-19:
@aparcar is this still relevant? Or maybe we can make this ticket more concrete.

mwarning, 2021-08-31:
not applicable anymore

closed 2021-08-31 by mwarning