mwarning/openwrt-firmware-selector #26: Improve image search behaviour

valerysntx, 2020-03-30:
Search field should  🔍 allow to **scroll,** and **filter by any containing symbols**

valerysntx, 2020-03-30:
Maybe regex here?
https://github.com/mwarning/yet_another_firmware_selector/blob/a646e40d9f830581c3e68c3af08469ee574c7c9c/index.js#L170

mwarning, 2020-03-30:
Scrolling works for me (up/down with arrow keys). Maybe you mean something different?

What do you mean by filtering by any containing symbols? You should be ableto search for non-ascii characters. As for regex support, I do not think that is a good idea. It would be too complex for the task.

valerysntx, 2020-03-31:
> Scrolling works for me (up/down with arrow keys)

Sorry, but its working only for desktop, right?  
> What do you mean by filtering by any containing symbols? 

 For example writing _tp-link_
 part of the name of device, while searching for needed model.

Maybe its really out of scope of your project now, just expected little more better experience.

mwarning, 2020-03-31:
> > Scrolling works for me (up/down with arrow keys)
> 
> Sorry, but its working only for desktop, right?
Yes, but I do not know how to scroll on mobile in general.
> 
> > What do you mean by filtering by any containing symbols?
> 
> For example skip writing _tp-link_
> part of the name of device, while searching for needed model.
I see. You like to search for two or more substrings at once.
That could be done without regexprs.
> 
> Maybe its really out of scope of your project now, just expected little more better experience.
Matching can be improved (multiple search terms, fuzzy matching). That's fine if it does not require too much extra code.

mwarning, 2020-06-19:
This is a rather complicated goal. Given the good enough state of the current approach, this is unlikely to be improved (unless someone comes up with a simple and elegant improvement).
Thank you for your input!

closed 2020-06-19 by mwarning