mwarning/openwrt-firmware-selector #39: Link only current download-files

FreifunkUFO, 2020-06-09:
when i build firmware several times in the same browser-window (with different additional packages) i get more and more download-icons. 
see screenshot, only the two files in the center can be downloaded

FreifunkUFO, 2020-06-09:
[image]

screenshot firefox

mwarning, 2020-06-09:
hi FreifunkUFO,

we would need to find out if:

1. this is a bug in the collect.py script
  - how to you call the collect.py script?
2. this is a bug in the openwrt generation of the JSON metadata
  -  can you check if the json files in the image folder have duplicated entries?

FreifunkUFO, 2020-06-09:
i am using firefox on https://chef.libremesh.org

the bug can be seen also from other computers. just open it, choose 902acv1 and write some package into "customize". you should now see two more download-icons then me.

maybe files are useful (all 902acv1-generated files) but are not named. (i am also missing the build-log to have a view)

[image]

mwarning, 2020-06-09:
hm, it works for me (Firefox 75.0). What browser version do you use?

mwarning, 2020-06-09:
nvm., I saw the same. Not sure how I did it..

mwarning, 2020-06-09:
Ok, this is the build server sending information of a bunch of images.
@aparcar I think you need to fix your build server.

aparcar, 2020-06-09:
Yes I found this problem too and here is the solution. https://github.com/openwrt/openwrt/pull/3091

mwarning, 2020-06-09:
Closing this issue here, as this is not a yafs bug.

mwarning, 2020-06-09:
thanks @FreifunkUFO for reporting!

closed 2020-06-09 by mwarning