mwarning/openwrt-firmware-selector #41: get rid of misc directory

FreifunkUFO, 2020-06-12:
the misc directory should be somewhere else, not it this project. 
it might be ASU for a good place, ( or might be there already?)

mwarning, 2020-06-12:
But where can I put the e.g. screenshot?

Is what you meant to say, that you want to do a checkout somewhere in production and update via "git pull"? I that case, misc would be in placed in production.

FreifunkUFO, 2020-06-12:
i criticise all files EXCEPT the screenshot
thereshould be no need for python or git on a device with YAFS

yafs is not able to know which openwrt-versions and hardware can be build by corresponding ASU-server.
its should be the task of ASU to provide these lists on demand.
(maybe ASU has this feature already?)

mwarning, 2020-06-12:
> i criticise all files EXCEPT the screenshot
The profiles.json files in misc are required for the demo. The python script is kind of a necessary evil unless OpenWrt creates an optimized single profiles.json file.

I also like this project to be as self contained as possible.

> thereshould be no need for python or git on a device with YAFS
I agree with git, of course. But a script is needed to merge the json files.
> 
> yafs is not able to know which openwrt-versions and hardware can be build by corresponding ASU-server.
> its should be the task of ASU to provide these lists on demand.
> (maybe ASU has this feature already?)
ASU support is an optional to YAFS and I do not want to have it as must have feature.
Otherwise this project has no use for me.

mwarning, 2020-06-15:
We can't get rid of this additional folder unless we want to spam the root folder.

closed 2020-06-15 by mwarning

mwarning, 2020-06-18:
The web content was now moved to a www folder. Maybe that addresses your concern to some degree.