mwarning/openwrt-firmware-selector #42: luci support

FreifunkUFO, 2020-06-12:
would be nice to have this project in luci
it is only html and js, so it should be possible.

but the misc-directory (incl. python) has to be changed first
https://github.com/mwarning/yet_another_firmware_selector/issues/41

FreifunkUFO, 2020-06-12:
a bit like https://github.com/openwrt/luci/tree/master/applications/luci-app-attendedsysupgrade , but also for build other-devices images or images for that device, but with different settings (f.i. packages or config or files)

mwarning, 2020-06-12:
I do not understand how the misc folder is a problem, because YAFS as LuCI module would be a new project that has different folders etc. anyway.

mwarning, 2020-06-15:
There is the luci-app-attendedsysupgrade (ASU) package. That should do it.

closed 2020-06-15 by mwarning