mwarning/openwrt-firmware-selector #44: usage for libremesh / chef

FreifunkUFO, 2020-06-12:
atm this project was claimed to be the official chef.libremesh.org, the successor of the former chef (https://github.com/libremesh/chef/) which had more features than this project atm.

please provide readme, documentation oder hints:
* how to create images for libremesh
* how to change package-list, flavors, network-config and local settings

mwarning, 2020-06-12:
> atm this project was claimed to be the official chef.libremesh.org, the successor of the former chef (https://github.com/libremesh/chef/) which had more features than this project atm.

I think there is a misunderstanding. ASU support was added to support libremesh/chef, but I need YAFS for Freifunk projects I am involved in. They do not use libremesh or gluon and the other firmware selectors available have a hard dependency on Gluon.

> 
> please provide readme, documentation oder hints:
> 
>     * how to create images for libremesh
This topic feels to be too far removed from this project...
> 
>     * how to change package-list, flavors, network-config and local settings
This should be documented in the ASU project, as it is their API.
YAFS should link to that.

closed 2020-06-15 by mwarning

FreifunkUFO, 2020-06-15:
ok i will ask libremesh people