mwarning/openwrt-firmware-selector #57: Swapping version after selecting model shows blank page

mweinelt, 2020-09-14:
Selecting a version and model shows the firmware. Swapping out the version gets you a blank page until you reconfirm the model, although the model string is still there.

mwarning, 2020-09-14:
Can you provide an example (and the link you were using)? I just tried "TP-Link TL-WDR4300 v1" and switched the version/release. The entry was uploaded.

mweinelt, 2020-09-14:
Ah, I tried the R7800. Between 19.07 and master the target name changed from `ipq806x` to `ipq806x-generic` IIRC, so that's a special case.

mwarning, 2020-09-14:
There were a lot of name changes (aka cleanup) between 19.07 and master. There is also the ar71xx to ath79 transition. Theses things are causing problems right now.

I particular, I see "Netgear Nighthawk X4S R7800"  19.07.4 and "NETGEAR Nighthawk X4S R7800" for master. The code does not use fuzzy matching when the version is switched. This is the problem here.

I will look into the problem later.

mwarning, 2020-09-23:
Fixed. The model should stay selected when a different version is selected (given the device id or device title stay the same between versions/releases).

closed 2020-09-23 by mwarning