mwarning/openwrt-firmware-selector #58: Update Translations

mwarning, 2020-10-11:
Four sentences were updated [1] from suggestions of the OpenWrt Admin Mailing List [2].
It would be great to have the translations updated.

[1] https://github.com/mwarning/yet-another-firmware-selector/commit/bc70ad1a0fd9fff3af543f8a3dd6c7f2522ed1f7
[2] https://lists.infradead.org/pipermail/openwrt-adm/2020-September/001678.html

mwarning, 2020-10-12:
You can also put the translations in here if you like.

castillofrancodamian, 2020-12-09:
> @rogerpueyo, @castillofrancodamian, @Dalen-ROP, @Ansuel, @obsy, @sinancetinkaya, @erdoukki please have a look - thank you!
> 
> Anyway, the code will copied to https://gitlab.com/openwrt/web/firmware-selector-openwrt-org and is currently deployed at https://firmware-selector.staging.openwrt.org/. An official deployment is imminent.

Where should I update the translation?

mwarning, 2020-12-09:
> Where should I update the translation?
One of the repos. It does not really matter which, as I will mirror commits to the other repo.

mwarning, 2021-01-05:
Closing for now. This will be an ongoing process.

closed 2021-01-05 by mwarning