mwarning/openwrt-firmware-selector #69: Feature: show x86-images

Akira25, 2021-03-06:
Though there are images for the x86-target, they don't appear in the selector. I assume, that the `profiles.json` files aren't generated for x86-images.

It would be really cool, to have them displayed in the selector too. I consider this an Issue of rather low priority, as there is a workaround available via the `Files`-Button in the right-down-corner

mwarning, 2021-03-06:
Hi @Akira25, yes, the profiles.json war not generated for the x86 targets. But it has been fixed for SNAPSHOT/20.02.
You can see it here: https://firmware-selector.openwrt.org/

Akira25, 2021-03-06:
Great. :) Then we should close this.

closed 2021-03-06 by Akira25

mwarning, 2021-03-06:
You can add it by hand, of course. But that is a bit of work..