mwarning/openwrt-firmware-selector #71: Improving the "info" link

pmelange, 2021-04-07:
@Akira25 and I discussed the firmware selector a bit and we have a recommendation for the info link.

The search for "TP-Link TL-WDR4900 V1" does not result in the device page showing up anywhere near the top of the search results.  But adding @toh to the search improves it a bit.

old info request URL https://openwrt.org/toh?do=search&id=toh&sf=1&q=TP-Link+TL-WDR4900+V1
new info request URL https://openwrt.org/toh?do=search&id=toh&sf=1&q=TP-Link+TL-WDR4900+V1%20%40toh

mwarning, 2021-04-07:
Thanks, I have added ` @toh` to the info link. It does not seem to make a different if I URL-encode the ` @toh`. The browsers translates it beforehand.

closed 2021-04-07 by mwarning

mwarning, 2021-04-07:
Btw., there is some progress of making the firmware selector and the wiki url compatible: https://forum.openwrt.org/t/the-openwrt-firmware-selector/81721/118