mwarning/openwrt-firmware-selector #73: Prevent browser from caching image build time

Akira25, 2021-04-27:
### Problem Description

In current Version 3.4.1 one can observe this behavior:
If the image-json-data got updated (i.e. via cronjob), the new image data doesn't show up in the browser. Apparently, its cached. I've notice that, when  I was loading a brand new snapshot-build, but the date was still pointing out that this image would have been built a month ago. Disabling the cache in development tools fixed that immediately.

### Expected Behaviour

It would be really great to have that caching somehow deactivated by default. Most users won't select that option in their browsers and would get displayed deprecated data then.

mwarning, 2021-04-28:
I think that would help: https://gitlab.com/openwrt/web/firmware-selector-openwrt-org/-/merge_requests/51

mwarning, 2021-04-29:
A fix was merged. Let me know if it works for you. Thanks!

Akira25, 2021-05-01:
Works fine. Thank you for that quick fix. :)

closed 2021-05-01 by Akira25